makerbase-mks/MKS-DLC32 #303: Sovol 40W Laser wiring

coldstorm82, 2022-12-12:
Hi Gentlemen,

I own the Sovol 40W diode laser (https://sovol3d.com/products/40-w-laser-engraver-kits-for-3d-printer-cnc-machine-laser-engraving-cutting-5w-output), but I have problems with wiring.

The laser has two two-pin connectors at the top, one marked with Power (+ -) the other with PWM (+ -). I couldn't figure out how to connect it to the MKS DLC32 V2. 

Currently I attached Power to the spindle-connector and I'm able to use the laser with different power settings.

But how do I connect PWM? To the TTL-pin? Or do I have to connect PWM to the spindle and take Power from another connector? I don't want to fry my laser ;-).

Thanks for your support!

manuelchaves, 2022-12-13:
Ok, on the board, upper right corner, you have a S-TTL-V connector where we have Signal +12/24V and -.Connect the - and the TTL to the respective connector of your SOVOL 40W, resolved the TTL signal. As for power, it is recommended to connect this S-TTL-V (+ 12/24V and -) to a MOS switching module, like this one, and using direct power from the power supply and never powered directly by the MKS DLC32. Was I able to help you?
[image]
[image]
[image]

coldstorm82, 2022-12-13:
Hi Manuel,

thanks a lot for your help!!!

Just to make sure I understood you well, I did a little sketch for the wiring. Cause I'm not quite sure about the connection of the MOS module. Do I also need to connect the PWM signal to the MOS?

(I studied mechanical engineering, I'm not so deep in the electrical stuff 😅)

[image]

Again, thanks a lot!

manuelchaves, 2022-12-13:
yes, correct

coldstorm82, 2022-12-13:
Then again, thanks for your support. 😊👍

closed 2022-12-13 by coldstorm82