makerbase-mks/MKS-DLC32 #307: Laser ui in cnc version

team-ajas, 2022-12-15:
Hello, I have installed the version V2.1.0_H24_20220511_01_N_CNC and the web ui does not correspond to cnc if not the ui for laser. Some help?
Thank you

DRF-Solutions, 2022-12-28:
Yeah, I can confirm this issue. Seems to be across the last few versions of CNC firmware releases. The LCD show's CNC tools, but the web UI shows laser controls - so of course, there's no Z-Axis controls in the web UI.

Also noted that there hasn't been a CNC version release in some time for the TS35, but laser firmware just very recently...

team-ajas, 2022-12-30:
Indeed, also in the lcd interface the z-probe icon is implemented and it is also missing in web ui. the frimware works correctly in cnc and the z probe from software is also only the web interface. thank you