makerbase-mks/MKS-DLC32 #309: Compiled firmware for wrong board

msy021, 2022-12-24:
The compiled firmware found at "firmware/TS35/Laser/Normal/Board_V1.0/V2.20_H35_2022_1101_N_ZX_002.bin" has the pins assigned as for v2 board and thus is unusable on v1 boards.
Please repair.

msy021, 2023-01-10:
Hello.
I have the board mounted on a pretty basic 2 axis laser engraver. The x and y motors and endstops work, but the laser output does not. I am not able to test other pins easily as the board is mounted in a casing and I am currently using the machine and do not have time for testing. 
Below is a screen capture showing the firmware version and falsely reported board version.
The board that I am using is V1 and the firmware image at "firmware/TS35/Laser/Normal/Board_V1.0/V2.0.8_H35_20220105_N.bin" works as expected. (motors, endstops and laser output)
[image]

abinoer, 2023-01-11:
Please use the firmware that matches the board, Board_V2.0

closed 2023-01-11 by abinoer

msy021, 2023-01-11:
But my board is V1.0. ...

msy021, 2023-01-11:
[image]
The firmware in the picture works ok on my board but it is an older version.