makerbase-mks/MKS-DLC32 #310: Can not start g-code from SD

Automatex09, 2022-12-24:
MKS-DLC32 V2.1 (printed on circuit board)
Display says: MKS DLC32 CNC V2.0 : V2.1(8M.H35.20220510) : CPU:Freq:240MHz/ T:53.3C/ ID:42017
As soon as I start a g-code program from the SD card, I get the error message: ERROR:1 IN SD FILE AT LINE 1
G code : G0 Z5.0000
If I enter the same code directly, it will be executed.

what could be the reason:

MKS-DLC32 V2.1 (Aufdruck auf der Platine)
Display sagt:MKS DLC32 CNC V2.0  :  V2.1(8M.H35.20220510) : CPU:Freq:240MHz/ T:53.3C/ ID:42017
sobald ich ein g-code programm von SD Karte starte bekomme ich die Fehlermeldung : ERROR:1 IN SD FILE AT LINE 1
G-Code : G0 Z5.0000
Gebe ich den Selben code direkt ein wird er ausgeführt. 

woran kann das liegen:

danielinkiel, 2022-12-24:
gibst Du händisch auch g0 z5"."0000 ein? Also den Punkt? Warum?

Automatex09, 2022-12-24:
Nein habe die Zeilen welche ich getetet habe kopiert und eingefügt, so das diese wirklich gleich sind 
Habe alle möglichen befehle getestet , und immer das gleiche ergebnis.

Automatex09, 2022-12-24:
Ich habe das Bord zwei mal, und bei beiden das gleich Problem.
Habe nun auf dem Zweiten Bord Fluid NC installiert, und da ist das gleiche, egal welche Befehle im CNC Programm sind, händisch funktionieren sie , als Programm gesendet nicht.
Ich bin am Verzweifeln !!

danielinkiel, 2022-12-24:
Womit hast Du oder WAS hat das "Programm" erstellt? g0.... ist da doch nicht die erste Zeile?!

Automatex09, 2022-12-24:
Da das eigentliche programm nicht funktioniert hat habe ich einfach g-codes ausprobiert, und keiner hat funktioniert.
Immer die gleiche Fehlermeldung.

danielinkiel, 2022-12-24:
najaa...dieses "Programm" wurde ja von irgendwas generiert (?) Postprozessor?

Automatex09, 2022-12-24:
Beispielprogramm:
(Projekt FamMetzger)
(Erstellt mit Estlcam Version 11 Build 11,244)
(Laufzeit ca. 00:06:43 Stunden)

(Benötigte Werkzeuge:)
(Fräser 0,3mm)

M03 S24000
G00 Z5.0000

(Nr. 1 Ausschnitt: Ausschnitt 14)
G00 X5.9386 Y5.4762
G00 Z0.5000
G01 Z0.0000 F500 S24000
G01 Z-0.2000

Automatex09, 2022-12-24:
Nachdem so nichts funktioniert hat ( auch einige andere programme) habe ich selber welche mit den einfachsten G-Codes geschrieben. Editor und abgespeichert mit .NC als Endung
Aber immer mit dem gleichen Ergebnis.

danielinkiel, 2022-12-24:
Da steht ja (ab ....Ausschnitt 14)), dass er zu x/y fahren soll. Dann kurz hoch und dann weiter.... ?!

Hattest Du mit demselben Postprozessor schon mal diese Abbrüche? bzw DAS mal mit alten Daten vergleichen? Ich fahr 3 Achsen mit nem Laser und einem zusammengebastelten Postprozessor. Bei mir steht da vorher viel mehr. zb woran sich das alles orientieren soll. G90 oder G54 usw

Automatex09, 2022-12-24:
Nun Ja , das ist nur der Anfang des programmes, die fehlermeldung bezieht sich ja immer auf die zeile 1. deshalb hab ich nicht das ganze Programm gepostet.
Auf dem DLC32 hab ich noch nichts am laufen, sind meine ehrsten versuche mit diesem Teil. 
Werde mal G90 und G54 versuchen bei meinem anderen system ( DDCS V2.1) brauche ich das nicht, da laufen diese Programme einwandfrei.Kann mir allerdings nicht vorstellen, das es daran liegt, denn er meldet ja immer es sei kein G- Befehl in Zeile 1

Automatex09, 2022-12-24:
Schade hat nichts gebracht, ich denke diese Befehle werden oft zur sicherheit vorab gesendet.Die meisten Steuerungen starten im absolut G90 modus und im arbeitskoordinatensystem G54.

Automatex09, 2022-12-24:
Hab nun den UGS installiert und damit getestet.
Wenn ich das gleiche Programm über UGS ( universal GCode Platform )sende leuft die Steuerung.
Hilft mir aber nicht weiter, da ich die Steuerung ohne Rechner verwenden will.

danielinkiel, 2022-12-24:
Verstehe. Ich lasse das tatsächlich immer über einen Rechner laufen.
Wenn du mir die ersten Zeilen schickst, schließe ich Montag Morgen (!) mal das tft an und probiers über mal. Allerdings wackelt der Vergleich noch, weil ich grad nicht weiß welche Firmware ich auf den dingern habe.

Automatex09, 2022-12-24:
Vielen Dank !!! das wäre super !!
Also hab Estlcam auf GRBL G-Code umgestellt und das Programm für ein Teil meines Gehäuses für den dlc32 erstellt.
Gleiches Problem  !
wäre Toll, wenn du das auf deinem testen könntest!
Ich denke ich habe irgendetwas in den einstellungen im esp32 falsch ?
Schöhne Feiertage !!

(Projekt GehauseOben)
(Erstellt mit Estlcam Version 11 Build 11,244)
(Laufzeit ca. 00:02:39 Stunden)

(Benoetigte Werkzeuge:)
(Fraeser 0,8,mm)
G21
G90
G94
M03 S24000
G00 Z5.0000

(Nr. 1: Ausschnitt 1)
G00 X8.3561 Y0.4150
G00 Z0.5000
G01 Z0.0000 F200 S24000
G01 Z-0.4000
G01 X6.6561
G01 Y6.4884
G03 X6.2561 Y6.8884 I-0.4000 J0.0000
G01 X5.3593
G01 Y7.6152
G03 X6.3341 Y7.6229 I0.4580 J3.7329
G03 X6.6561 Y8.0152 I-0.0780 J0.3923
G01 Y14.7604
G01 X8.3561
G01 Y8.0152
G03 X8.7561 Y7.6152 I0.4000 J0.0000
G01 X9.6021
G01 Y6.8884
G03 X8.6781 Y6.8807 I-0.4326 J-3.5400
G03 X8.3561 Y6.4884 I0.0780 J-0.3923
G01 Y0.4150
G00 Z5.0000

(Nr. 2: Ausschnitt 2)
G00 X63.2537 Y6.8884 Z5.0000
G00 Z0.5000
G01 Z0.0000 F200
G01 Z-0.4000
G01 X62.3569
G01 Y7.6152
G03 X63.3317 Y7.6229 I0.4580 J3.7329
G03 X63.6537 Y8.0152 I-0.0780 J0.3923
G01 Y14.7604
G01 X65.3537
G01 Y8.0152
G03 X65.7537 Y7.6152 I0.4000 J0.0000
G01 X66.5997
G01 Y6.8884
G03 X65.6757 Y6.8807 I-0.4326 J-3.5400
G03 X65.3537 Y6.4884 I0.0780 J-0.3923
G01 Y0.4150
G01 X63.6537
G01 Y6.4884
G03 X63.2537 Y6.8884 I-0.4000 J0.0000
G00 Z5.0000

(Nr. 3: Teil 1)
G00 X72.0000 Y-0.4000 Z5.0000
G00 Z0.5000
G01 Z0.0000 F200
G01 Z-0.4000
G03 X72.4000 Y0.0000 I0.0000 J0.4000
G01 Y20.5600
G03 X72.2488 Y20.8732 I-0.4000 J0.0000
G01 X49.9590 Y38.5781
G03 X49.8113 Y38.6519 I-0.2488 J-0.3132
G01 X43.1365 Y40.3950
G03 X42.6777 Y39.8290 I-0.1011 J-0.3870
G01 X55.7170 Y13.7683
G01 X54.3973 Y13.1824
G01 X34.8558 Y52.1083
G03 X34.4983 Y52.3288 I-0.3575 J-0.1795
G01 X0.0000
G03 X-0.4000 Y51.9288 I0.0000 J-0.4000
G01 Y0.0000
G03 X0.0000 Y-0.4000 I0.4000 J0.0000
G01 X72.0000
G00 Z5.0000
G00 X0.0000 Y0.0000

M05
M30

danielinkiel, 2022-12-26:
Moin

Jo. Läuft. über den PC UND SD-Karte mit Firmware wie auf dem Screenshot

Hab "S" wie folgt geändert und einfach laufen lassen:

G21
G90
G94
M03 S5
G00 Z5.0000

(Nr. 1: Ausschnitt 1)
G00 X8.3561 Y0.4150
G00 Z0.5000
G01 Z0.0000 F200 S5
G01 Z-0.4000
G01 X6.6561
G01 Y6.4884
G03 X6.2561 Y6.8884 I-0.4000 J0.0000
G01 X5.3593
G01 Y7.6152
G03 X6.3341 Y7.6229 I0.4580 J3.7329
G03 X6.6561 Y8.0152 I-0.0780 J0.3923
G01 Y14.7604
G01 X8.3561
G01 Y8.0152
G03 X8.7561 Y7.6152 I0.4000 J0.0000
G01 X9.6021
G01 Y6.8884
G03 X8.6781 Y6.8807 I-0.4326 J-3.5400
G03 X8.3561 Y6.4884 I0.0780 J-0.3923
G01 Y0.4150
G00 Z5.0000

(Nr. 2: Ausschnitt 2)
G00 X63.2537 Y6.8884 Z5.0000
G00 Z0.5000
G01 Z0.0000 F200
G01 Z-0.4000
G01 X62.3569
G01 Y7.6152
G03 X63.3317 Y7.6229 I0.4580 J3.7329
G03 X63.6537 Y8.0152 I-0.0780 J0.3923
G01 Y14.7604
G01 X65.3537
G01 Y8.0152
G03 X65.7537 Y7.6152 I0.4000 J0.0000
G01 X66.5997
G01 Y6.8884
G03 X65.6757 Y6.8807 I-0.4326 J-3.5400
G03 X65.3537 Y6.4884 I0.0780 J-0.3923
G01 Y0.4150
G01 X63.6537
G01 Y6.4884
G03 X63.2537 Y6.8884 I-0.4000 J0.0000
G00 Z5.0000

(Nr. 3: Teil 1)
G00 X72.0000 Y-0.4000 Z5.0000
G00 Z0.5000
G01 Z0.0000 F200
G01 Z-0.4000
G03 X72.4000 Y0.0000 I0.0000 J0.4000
G01 Y20.5600
G03 X72.2488 Y20.8732 I-0.4000 J0.0000
G01 X49.9590 Y38.5781
G03 X49.8113 Y38.6519 I-0.2488 J-0.3132
G01 X43.1365 Y40.3950
G03 X42.6777 Y39.8290 I-0.1011 J-0.3870
G01 X55.7170 Y13.7683
G01 X54.3973 Y13.1824
G01 X34.8558 Y52.1083
G03 X34.4983 Y52.3288 I-0.3575 J-0.1795
G01 X0.0000
G03 X-0.4000 Y51.9288 I0.0000 J-0.4000
G01 Y0.0000
G03 X0.0000 Y-0.4000 I0.4000 J0.0000
G01 X72.0000
G00 Z5.0000
G00 X0.0000 Y0.0000

M05
M30

Einstellungen grbl auf dem Board wie folgt (wie schon geschrieben: (Dioden)LASER):
$0=10 (Step pulse time)
$1=25 (Step idle delay)
$2=0 (Step pulse invert)
$3=7 (Step direction invert)
$4=0 (Invert step enable pin)
$5=1 (Invert limit pins)
$6=0 (Invert probe pin)
$10=1 (Status report options)
$11=0.010 (Junction deviation)
$12=0.002 (Arc tolerance)
$13=0 (Report in inches)
$20=0 (Soft limits enable)
$21=0 (Hard limits enable)
$22=1 (Homing cycle enable)
$23=3 (Homing direction invert)
$24=150.000 (Homing locate feed rate)
$25=1100.000 (Homing search seek rate)
$26=35.000 (Homing switch debounce delay)
$27=1.000 (Homing switch pull-off distance)
$28=1000.000 ()
$30=1000.000 (Maximum spindle speed)
$31=0.000 (Minimum spindle speed)
$32=1 (Laser-mode enable)
$38=0 ()
$40=1 ()
$100=80.340 (X-axis travel resolution)
$101=80.100 (Y-axis travel resolution)
$102=399.940 (Z-axis travel resolution)
$103=100.000 ()
$104=100.000 ()
$105=100.000 ()
$110=4000.000 (X-axis maximum rate)
$111=4000.000 (Y-axis maximum rate)
$112=2000.000 (Z-axis maximum rate)
$113=1000.000 ()
$114=1000.000 ()
$115=1000.000 ()
$120=100.000 (X-axis acceleration)
$121=100.000 (Y-axis acceleration)
$122=80.000 (Z-axis acceleration)
$123=200.000 ()
$124=200.000 ()
$125=200.000 ()
$130=550.000 (X-axis maximum travel)
$131=750.000 (Y-axis maximum travel)
$132=160.000 (Z-axis maximum travel)
$133=300.000 ()
$134=300.000 ()
$135=300.000 ()

[image]
[image]
[image]

Automatex09, 2022-12-26:
Hallo,
Vielen Dank !
Ich kann nicht sagen wieso, aber nachdem ich die firmware neu aufgepielt habe funktioniert es jetzt auch bei mir !!
Allerdings denke ich das es wirklich mit dem Postprozessor, welcher nicht auf grbl eingestellt war zusammen hing.
Denn nun funktioniert es auch auf dem Bord mit fluidNC!
danke noch einmal !!
gruß Alex

danielinkiel, 2022-12-26:
aye

closed 2022-12-26 by Automatex09