makerbase-mks/MKS-DLC32 #311: TTL laser wiring

mslmkytz, 2022-12-26:
hello the laser i use is ATOMSTACK A5 PRO 40W
TTL output 2-wire + .- single cable. DLC32 V2.1 I am using
How should the cable assembly of my TTL output 3-wire laser be?

manuelchaves, 2022-12-28:
Recommended is to share the GND with the laser and the TTL and the 12/24V output and the GND trigger a power MOSFET driver, this way you preserve the output of the DLC32 like these
[image]
[image]

abinoer, 2023-01-07:
> Recommended is to share the GND with the laser and the TTL and the 12/24V output and the GND trigger a power MOSFET driver, this way you preserve the output of the DLC32 like these [image] [image]

Yes, this connection is recommended

closed 2023-01-07 by abinoer