makerbase-mks/MKS-DLC32 #313: Firmware does not work with MKS TS35-R V2.0 monitor

latenikBR, 2023-01-02:
MKS TS35-R V2.0 monitor does not work with firmware 2.20. I had to downgrade to version 2.10 for it to work again. Any solution?

abinoer, 2023-01-07:
Do you mean that you cannot burn 2.2+ firmware? If so, you need to set the burning setting "FLASH SIZE" to 8MB

latenikBR, 2023-01-18:
I mean: The MKS TS35-R v2.0 screen which is sold in a kit together with the MKS-DLC32 board. I said "monitor" but I meant "screen". In my country is very similar words. Sorry. I mean the screen turns off when installed firmware v2.20. But it works again in version 2.10. I do not know why.

abinoer, 2023-02-27:
Hello, the screen is universal. Logically speaking, there is no such thing as what you said. Please confirm again whether the firmware has been burned successfully. You can send pictures or videos here. (Note that the 2.2+ firmware requires the flash size to be set to 8MB)

Lucan07, 2023-10-20:
The character set encoding used on MKS T35 v2.0 is no longer supported cannot find on marlin site that still claims screen is supported MKs offering great deals that Marlin will not support anything after 2.09 something. so MKS dum off in sales while keeping it quiet, its on Marlin site but could not find again a moment ago.