makerbase-mks/MKS-DLC32 #317: MKS DLC32 V2.01 Firmware?

targor, 2023-01-05:
Which firmware do i need to use when i want to update my MKS DLC32 V2.01?

- There is TS24 and TS35, and i cannot see anywhere which one applies to my board. I could guess it stands for "Touch screen", but i even have not applied one.. 

- Also, .. Laser or Core X/Y?

I want to use a pwm led laser on this board.

Some more descriptions what all the trXX, Core XY things are for would be very great.. 
Please help, i am lost

nofuturekid, 2023-01-05:
If you do not have a touchscreen attached, then it doesn't matter if you use TS24 or TS35.

If you need the 'normal' or 'CoreXY' *we* can't tell you, because of lack of information as our glass spheres are in maintenance and you did not tell us on what machine you have mounted that board. 

But very likely you need the 'normal' one, if you ask that kind of question.

targor, 2023-01-05:
Thanks for the info. Maybe you could update the readme that it does not matter to use TSxx if someone does not have a display at all.

My question would be ... For what kind of device would the CoreXY be used for?

Or the other way around.. if i would write a readme.. : If you have a xy device, please use the CoreXY firmware, if not, use the normal one

nofuturekid, 2023-01-05:
Looks like don't know what CoreXY is all about...
It is a way how the mechanics moves the laserhead around. It's mostly used in 3D printing. So whoever uses CoreXY usually knows he does.

I'm not part / involved in that project, at least not more than you. So your idea of writing these additional notes yourself is a good one.

manuelchaves, 2023-01-05:
I will try to help you:
->TS24 and TS35 are the display sizes, if you purchased with it. There are differences in icons and information displayed due to size.
->The coreXY is a pattern of how the X and Y axis are activated, contrary to the traditional Cartesian pattern, see the images.
-> First you determine if your TS24 or TS35 display, if you don't have a display, either one answers. Then laser, and the type of movement of its axes
[image]
Corexy

[image]
[image]

targor, 2023-01-05:
Thank you for all the help. i updated the readme myself now and created a pull request in hope, that it gets trough.

closed 2023-01-05 by targor