makerbase-mks/MKS-DLC32 #320: Firmware v2.30 WIFI STA connect issue

nofuturekid, 2023-01-07:
After flashing v2.30  switch from AP mode into STA mode and connect to the local network works fine.
As soon the board is powered off and on again, it will not automatically re-connect the access point.
I have to go to config/wifi and then it will automatically "scan" and connect again.

Please fix, so that once it's switched into STA mode it will automatically reconnect when power up again.

IronDuke123, 2023-01-07:
X2 on this, it never auto connected for me even with the older firmware. Would be nice if it did.  The web config for this failed for me as well, had to do it with the little screen on the controller.

AndWalu, 2023-01-07:
Auto connect in STA mode is important to me too.
Is it possible to implement it in next update?

nofuturekid, 2023-01-07:
> X2 on this, it never auto connected for me even with the older firmware. Would be nice if it did. The web config for this failed for me as well, had to do it with the little screen on the controller.

At least v2.10 did autoconnect for me since v2.2 it is not working.

Thabke, 2023-01-13:
Have same issue. After flashing v2.30 firmware DLC32 begin to reset WiFi configuration after each restart.

abinoer, 2023-01-29:
Automatic WiFi reconnection causes some bugs.- Automatic WiFi reconnection has been removed, as mentioned in readme

TimoFeick, 2023-02-24:
Anyone know a workaround for Auto-Connection in STA mode?
Wireless Connection was the reason for my purchase...

AndWalu, 2023-02-24:
> Anyone know a workaround for Auto-Connection in STA mode?
> Wireless Connection was the reason for my purchase...

Mine too unfortunately...
Makerbase-mks do not provide the firmware source code so we can't do anything about Wi-Fi.
We can only ask FluidNC to add a LCD support to their software.

danielminson, 2023-03-11:
I also have this problem...

iherbak, 2023-06-15:
I stucked with the 2.8 firmware. it is the most stable for me so far and also no issue with wifi. Maybe give it a go

Bespinuk, 2024-08-30:
Did this ever get fixed? would be useful not to have to do this each time.

Gzikic, 2025-01-17:
Is it fixed "- Temporarily cancel automatic WiFi reconnection"?

francoissouza, 2025-04-24:
Hi, guys.
Has anyone found a solution? I'm facing the same problem.