makerbase-mks/MKS-DLC32 #322: H-Bot

FireSale14, 2023-01-09:
Hi,
kann ich dieses Board auch für einen H-Bot Antrieb nehmen? Leider hab ich nichts dazu gefunden.

Vielen lieben Dank
Grüße
Michael

danielinkiel, 2023-01-09:
Keine Hilfe, aber... wieder was gelernt (wtf is H-Bot? ;). 

Scheinbar nehmen die alle corexy firmware zb der hier:

[https://www.youtube.com/watch?v=K0pcGKAQ4eA&ab_channel=VangelisGavalakis
]

aller schwenken die auch schnell zu corexy um.

Was Dein Plan?

Grüße

Daniel

FireSale14, 2023-01-10:
Hi @danielinkiel ,
du bist der Beste. Ich kann jeden Controller der CoreXY kann für meinen H-Bot nehmen. Der Link war sehr hilfreich.

Ich habe einen SparkMaker LaserPro. Der wird aber nicht mehr weiterentwickelt und der Controller funktioniert nicht so toll mit Lightburn. Deswegen suche ich einen neuen Controller.

Grüße
Michael

danielinkiel, 2023-01-10:
"...funktioniert nicht so toll"?

FireSale14, 2023-01-10:
Der Rahmen mit Laser AN funktioniert nicht. Die Rotationsachse lässt sich nicht richtig einstellen usw. Der Originalcontroller speichert die Steps nicht richtig.

danielinkiel, 2023-01-10:
okokverstehe ;)

Der hier

https://www.youtube.com/watch?v=jlRYsr-DPnw&t=965s&ab_channel=MakersMashup

macht das mit nem skr board und Marlin. scheint zu laufen. Die Boards sind auch günstig.