makerbase-mks/MKS-DLC32 #324: Air Assist and Probe

concuus, 2023-01-11:
Hello, I have installed the latest version of COREXY TS35 on my DLC32 V2.1 card.
The system works without any problems, but I have a problem when I install air assist and probe.
Air Assist machine starts to work as it works, the probe seems to be triggered even though it is not triggered, how can I change them?

3Dangel, 2023-01-12:
> Hello, I have installed the latest version of COREXY TS35 on my DLC32 V2.1 card. The system works without any problems, but I have a problem when I install air assist and probe. Air Assist machine starts to work as it works, the probe seems to be triggered even though it is not triggered, how can I change them?
> 
> Edit: E-Stop also works in reverse

may be it is the config in Grbl that you might be missing?

concuus, 2023-01-12:
you're right, but I couldn't find enough documentation in my language. If there is a link that shows the config settings step by step, let me try again.

closed 2023-01-13 by concuus