makerbase-mks/MKS-DLC32 #325: Probe Target Flame Error

concuus, 2023-01-13:
Hello, first of all I use translation. Sorry if there is any confusion. I'm either missing something or doing something wrong as there isn't enough documentation in my native language, any help would be appreciated.
I am trying to make a CoreXY diode laser. For this, I use SculpfunS9 laser with MKS DLC32 V2.1, TS35 and TMC2209 set.
Since the bench I assembled is not on the scale, I installed the necessary swich and probe to measure with the probe.
I measure with the "G38.3 Z70 F200" command, but when the probe is triggered, I get the following error;

G38.3 Z77 F200
[MSG: Found]
[PRB:450,000,450,000,000,70,086:1]
arrow
[MSG:Flame error! Please restart the machine]

As far as I researched and understood; The $37 command turns the fire sensor on and off. However, the code 37$ appears to be disabled in the assembly numbered V2.30_H35_2022_1223_C_ZX_001 that I installed

37 dollars
error:3
The grbl '$' system command is not recognized or supported.
$37=0
error:3
The grbl '$' system command is not recognized or supported.

I would be glad if you could help me on the matter.

Additionally, I use Air Assist. Related Pin connections are correct but the system is working in reverse. It opens with M9 and closes with M7. Is there a Gcode that will reverse this cycle?

Thanks in advance.

concuus, 2023-01-19:
I guess this didn't happen to anyone.

MJ666, 2023-02-17:
I have the same error when trying this firmware with an Atomstack x20. This is using an LaserBox ESP32 V1.0 board which is likely a clone of the MKS DLC32.  If i use the prebuild firmware for an MKS V1.0 board everything is working except the laser does not fire. If I use an firmware fro the MKS V2.0 board i get this error "[MSG:Flame err!Please restart the machine]" and the board looks up after a few seconds and does not respond anymore. I found no way how to disable or control the fuction of the probe pin. The only difference between MKS V1.0 and V2.0 boards are the probe and spindle (laser pwm) pins. I try to use this firmware since the Atomstack firmware does not support M9, M8 and M9 commands to control the I2C SCL pin.

abinoer, 2023-02-27:
@concuus 
1. Warning about FLAME: The laser firmware has no probe function, because the io used for probe detection is used for flame detection in the laser firmware

2. About air assistance: M7————scl pin high level, M9——————scl pin low level, so it is necessary to choose NMOS module as the switching module

closed 2023-02-27 by abinoer