makerbase-mks/MKS-DLC32 #332: Inaccurate. Bad GCODE repeatability

villamany, 2023-01-27:
After a lot of headaches checking and rechecking all the hardware and almost giving up. I have verified that there is a serious bug in the firmware.
If you repeat the same gcode file over and over again you will see that the position of the engraving (i use a laser diode) varies very slightly making it impossible to finish a quality job that requires several passes (tenths of a millimeter).
There must be an error in the code, perhaps with the decimals that makes the repeatability bad.

Afte check-recheck all hardware, as a last resort I replaced the firmware by anothe one (fluidNC) and the result is perfect, as expected. All the passes match perfectly.

Which has led me to verify that it is a firmware bug

villamany, 2023-02-10:
Finally solved by increasing the Step pulse time from 6 to 8us. It seems to work fine. So i leave it at 10us for marging
Not much sense because fluidNC at 4us works fine, but at least now is working as expected.

prosoftJit, 2023-02-12:
I dont believe that. I have fighted this problem for some time now and cant find a solution. Always think its my hardware. Changed some parts... i just cant get two passes one exactly over another with my diy diode laser (zbaitu 20W). I made the change you suggested and will run some testes before report back... thank you

Charus-OV, 2023-02-16:
Yes. I can confirm. I had the same problem. We need to increase $0=10 and it will work well.

maadmackz, 2023-02-27:
i had to take step pulse to 10 , i also was noticing alot of interference in my wiring, after a more appropriate cleanup and cable management now it works as expected.