makerbase-mks/MKS-DLC32 #333: No Z-Axis available for Laser

2nice4u, 2023-01-30:
I updated the Laser V2.1 to the latest V2.3 and now there is no option to move the z axis!
Now I have no way to move the height-adjustable table.

[image]

philvtx, 2023-02-15:
Bonjour,
Oui j’ai le même problème mais surtout je n’ai qu’ 1 axe qui prends les références x ou y le plus près du switch !
Le Z aucune référence ne se fait ! J’ai flashé toutes les versions et toujours le même problème ! 
Carte MKS DLC32 ET LIGHTBURN

abinoer, 2023-02-27:
Yes, I'm sorry, this version does not plan Z-axis control

2nice4u, 2023-03-10:
> Yes, I'm sorry, this version does not plan Z-axis control

But I'm not the only one with a height-adjustable table via a stepper motor. The versions before that had offered this possibility. I see this as a step backwards.

cyberjak2k, 2023-06-03:
I agree this was a downgrade , we need the option of Z axis control otherwise board is not useful

cyberjak2k, 2023-06-03:
> Yes, I'm sorry, this version does not plan Z-axis control

can this please be revisited, this version fixed issue with compatibility with mobile app but removed Z axis control REALLY need this back or it means board no longer fit for purpose and might require return which I'd rather not do as I liked the idea of it

Robhoski, 2024-05-18:
Agree! I use the MKS DLC32 V2.1 to run my plasma cutter. I NEED Z_AXIS! The workaround sucks - gotta sero the z-axis in software first. PIA

cyberjak2k, 2024-05-18:
Yes it was removed in a previous version I agree this should be restored

thomonz, 2024-11-30:
Miss the z axis gor the laser version. Can this be re instated

cyberjak2k, 2024-11-30:
I think only by reverting to older version which I agree is a downfall