makerbase-mks/MKS-DLC32 #334: ESP32 bootloader_dio_80m.bin and Flash Download Tool v.3.9.3 problems

gbconnet, 2023-02-01:
With this software version (selecting chipType "ESP32" and workMode "develop"), using bootloader_dio_80m.bin@0x1000, default_8MB.bin@0x8000 and MyAPP.bin@0x10000 (SPI SPEED "80MHZ", SPI MODE "DIO"), the "CombineBin" button creates a target.bin file that does not work. I have found that the byte at address 0x1003 is 0x2F. 
With version 3.8.5 and earlier the generated target.bin file works, because the byte at address 0x1003 is 0x3F. This is the only difference.
Why? How can I set "Flash Download Tool" for generate the correct target.bin file?

gbconnet, 2023-02-14:
Can anyone help me please?

iherbak, 2023-02-16:
According my research and usage. The board is using dout and 40Mhz crystal. Please try to use bootloader_dout_40m.bin

gbconnet, 2023-02-16:
I tried to use bootloader_dout_40m.bin and also bootloader_dout_80m.bin, but ESP32 doesn't work. This is the output:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
ets Jun  8 2016 00:22:57

...

iherbak, 2023-02-16:
Have you tried vscode to upload?

gbconnet ***@***.***> (időpont: 2023. febr. 16., Cs, 15:28)
ezt írta: