makerbase-mks/MKS-DLC32 #335: Mid dlc32 v2.1 laser is always on

danielfrausto, 2023-02-01:
Help. My flashed the laser corexy firmware and laser is always on. Constant power. Setting $32=0 won't turn off

rd636, 2023-02-08:
I have the same issue.  The 12-24V-PWM type (2PIN) is constantly producing 12 volts (or 24 volts when using that power input voltage) immediately at startup and not turning off.   Nothing else seems to be impacted.  Drives motors fine.  Homing is still working.  LCD working.

danielfrausto, 2023-02-08:
> I have the same issue.  The 12-24V-PWM type (2PIN) is constantly producing 12 volts (or 24 volts when using that power input voltage) immediately at startup and not turning off.   Nothing else seems to be impacted.  Drives motors fine.  Homing is still working.  LCD working.  

Hello there, I was able to fix this issue by sending the commands 
M4 S0
Then
M5
And finally 
M4 S0 again.
Hope this works for you too.

rd636, 2023-02-08:
Thank you for the suggestion but it did not resolve the issue.   
M4 S0
ok
M5
ok
M4 S0
ok

villamany, 2023-02-09:
what pins are you using?. PWM is TTL level not 12/24v

rd636, 2023-02-10:
I think the board is bad.  I replaced the board, upgraded it, and am not experiencing the issue.   I think we can close this out.

maadmackz, 2023-02-11:
im having this to just suddenly, it was fine and then i hit buffer mAX it froze and now laser on all th8e timje, s to in,  ttl to L , gonna try that with spindle - to ground?  and external psu grounded to lpsu both grounded to case. both grounded to AC ground.

closed 2023-02-27 by abinoer

sharknetwork24, 2023-06-07:
I have the same problem too.
Everything works perfectly except that the laser is always on and to turn it off I have to disconnect the general power supply.
I would like to know if the problem has been solved, or if it is a defect of the Board DLC32 v2 that does not work.
Can it be solved or should the BOARD be replaced?
Thanks for the replies.