makerbase-mks/MKS-DLC32 #336: Wifi connection disconnected after restarting

SulzRohr, 2023-02-05:
Configuration: MKS DLC32 V2.1 with MKSTS35-R V2.0. Firmware :V2.30_H35_2022_1223_N_ZX_001.bin
Every time the board is restarted, the wifi connection must be reactivated. The IP address is retained.

Thabke, 2023-02-05:
In last update notes they mention that:
- Temporarily cancel automatic WiFi reconnection

closed 2023-02-11 by SulzRohr

tjivanhoff, 2023-09-06:
Hey boys hurry up. It's a fundamental feature. Nobody can say the opposite.
Even the worst of my smart plugs auto connects to wifi. There are no reasons why a DLC-32 can't.

maarato, 2024-10-08:
if anyone is having this issue you can solve this by installing next version:
https://github.com/makerbase-mks/MKS-DLC32/blob/main/MKS-DLC32-main/firmware/TS24/Laser/Normal/Board_V2.0/V2.0.8_H24_20220107_N.bin

this is for MKS DLC32 V2.1 also, seems like firmware inside TS35 folder have disabled the wifi auto connect.