makerbase-mks/MKS-DLC32 #340: About Spindle Control

mikaildogru, 2023-02-11:
Hey,

I have planning to make CNC with MKS Dlc 32. When ı research something about this, I have founded 0-5 to 10V 
[image]
İs there any experince in that card 

[image]

Can you share your experience with me ?

kalazzerx, 2023-03-18:
I am planning on using the setup with my vfd.  hope to have it working this weekend.  did you ever make it work?  I have not figured out all details yet - the correct pins to use, and gcode needed is what I am mostly trying to figure out.  The PWM to Voltage module is fairly straightforward forward and saw it was used by others with controllers.

Essentialartworks, 2023-07-02:
[image]

Got my VFD just now, any help will be appreciated