makerbase-mks/MKS-DLC32 #341: Another M4 error

A-J-Urso, 2023-02-18:
I’ve read all the previous info on the M4 error but I’m both new and have a Frankenstein rebuild.  I’m very new to laser engravers, controllers, and associated software.  I jumped in to help my friend because I operated CNC mills, programmed computer controllers in a past life, and did a lot of wiring.  

The base product was a wainlux jl3 that went bad for my friend he asked for help, the controller stopped moving the motors then the laser stopped lighting.  I replaced both parts and it runs but, has the M4 error issue.  

Current configuration is Lasertree 40 watt laser and MKS DLC32 V2.  I use lasergrbl for software and I’m in the trial period of Lightburn.  Lasergrbl burns but Lightburn doesn’t turn the laser on at all.  I have a TS-35 enroute so I can set up WIFI for we access.

Current problem appears to be laser power.
The laser will turn on and burn at max power but obviously doesn’t alter power when changing directions, since burn spots are obvious.  If I type the command M4=1 I get the laser must be enables error.
 
I’ve tested wiring with a multi-meter and board to laser connection is correct.  The laser will show low power when I hit lasergrbl focus button so I know it can be controlled.  I don’t know what else to do for diagnostics but from reading previous threads, I see the following on startup, which is promising, I think:  “MSG:PWM spindle Output:GPIO(32), Enbl:None   , Dir:none…….."

Any help in pointing me in the right direction is appreciated.  Remember I am new to this, so excuse stupid questions.

abinoer, 2023-02-27:
What's your specific question? Unable to control the laser output with lightburn, which can be controlled normally with lasergrbl?

A-J-Urso, 2023-02-27:
After building the Franklaser, it seems to work well with black and white drawings.  Grayscale does not appear to work at all with Lasergrbl or my trial version of Lightburn.  

Both the laser and control board have PWM capabilities, I understand laser intensity is used for grayscale, laser power is controlled by this pin, and that the M4 setting is what turns it on.  When I issue an M4 in lightburn or lasergrbl I get the above error.  Am I misunderstanding or do I have to fix something.