makerbase-mks/MKS-DLC32 #342: MKS get stuck while running G4

PashMato, 2023-02-19:
hi, I'm working on a little project and I'm trying to make the CNC machine to stop for a second after every command without any success. I'm using the `G4 P1` command (I guess it's in seconeds) some times (randomly) the MKS just stop moving
though the debug console is still running. this problem however doesn't happend the I don't use the `G4 P1` command

the console when the program get stuck:
`<Idle|MPos:-17.500,-7.287,4.225|FS:0,0|SD:6.50,/ship-circles.gcode>
<Idle|MPos:-17.500,-7.287,4.225|FS:0,0|Ov:70,100,100|SD:6.50,/ship-circles.gcode>`

my config:
`$0=10
$1=25
$2=0
$3=1
$4=0
$5=1
$6=1
$46=10.000
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=1
$24=300.000
$25=1000.000
$26=250.000
$27=1.000
$28=1920.000
$30=10000.000
$31=0.000
$32=0
$38=1
$40=1
$100=80.000
$101=80.000
$102=400.000
$103=100.000
$104=100.000
$105=100.000
$110=6000.000
$111=6000.000
$112=6000.000
$113=1000.000
$114=1000.000
$115=1000.000
$120=500.000
$121=500.000
$122=500.000
$123=200.000
$124=200.000
$125=200.000
$130=450.000
$131=450.000
$132=50.000
$133=300.000
$134=300.000
$135=300.000`

does any one have any ideas what's wrong? or is it a bug?

abinoer, 2023-02-27:
The G4 PXXX directive is available. You're saying that's an inaccurate pause? Is it random?

PashMato, 2023-02-27:
First, by the gcode manual `G4 P` in milliseconds but in reality it's in seconds. 

Second, the command feel random; sometimes the pause is for a long time and sometimes for a short time.

But the real deal breaker is that gcode get stuck randomly while running, that doesn't happened without the waiting 

Sorry for me being unclear