makerbase-mks/MKS-DLC32 #344: USB port is not responding :(

Rob208, 2023-02-23:
Laser Longer Ray5 using version DLC32 V003 -  Problem with USB 🙁.
In the midle of job laser stoped... and since then there is no communication via USB.
Alredy done:

- reset all the stuff (comp, laser etc...)
- checked on windows and macOS - windows shows that is blocking USB device because discovered problems, macOS just switched off the port
- on the laser you can see if throu the usb port - there is red diode inside light (look at the picture) and green... previously it was only green

Is it mean that motherboard for USB port is broken ???
Is there any chance to fix it ?

abinoer, 2023-02-27:
USB port not responding? Can't reburn firmware? Most likely the motherboard is damaged

TimE83, 2023-12-24:
I'm having exactly the same issue. Was it the board/hardware failure?

Deun67, 2024-05-04:
have you found a solution?

8tabbie, 2024-05-04:
No. Ended up buying a few more to have that works.

Kind regards
Louis Nel
071-3128581
Eastern Blue Investments
37 Oceanway, Gonubie