makerbase-mks/MKS-DLC32 #345: With SD card no conection via browser

andre1000, 2023-02-23:
Hello, I have a problem with my DLC32 v2. I have the actual firmware (v2.30).
When I have no SD-card inside, than I can connect the DLC with my browser. When I put the SD-card inside, than no connection is possible. The MKS-DLC32 is working. Display is on and I can also connect the APP. The SD-card is 16GB FAT32 or 8GB FAT32.
Any idea?

abinoer, 2023-02-27:
There is no such situation on my side, nor have I received similar feedback. Can you change the firmware to make normal connection? Maybe you can try two TF cards instead

andre1000, 2023-03-09:
I have tested 3 different types of SD-Cards 256MB, 8GB and 16GB. -> The same failure
I have tested the firmware 2.0.8/2.1.0/2.3.0 -> The same failure
I have found one additional point. When I use the engrave button on the display "No SD Card"
It is possible, that I have a defect on my board?

librarymark, 2023-03-19:
I am facing the same issue. I am able to ping the unit, but when I have any SD card in it, the webpage will not load. This is a bug that needs to be fixed.

costycnc, 2025-02-10:
I think that is problem is because the program try to find dlc_cfg.txt and if find or not the program run another way that blocking the program. Also i see that if have dlc_cfg.txt change $ESP115 from on to off and make off wifi , i think that is important to insert also the command [ESP115] in dlc_cfg.txt

costycnc, 2025-11-09:
Try reinstalling the firmware or installing a different version. Some firmware versions have bugs, but so far nobody has taken action to fix them