makerbase-mks/MKS-DLC32 #346: Air Assist, GRBL version in LightBurn and Z with rotary roller

manuelchaves, 2023-02-24:
Hi.
I have some questions:

- Which pin I use to connect the air assist?

- LightBurn, I configure it as GRBL, GRBL-LPC, GRBL-M3 or another controller?

- Its possible use the Z axis pin, without function until now, to drive the rotary roller?

Grateful

TimoFeick, 2023-02-24:
You can use G and SCL. I use an SSR with low voltage trigger (SCL delivers only 3.3v).
In my case a SSR-40DA with an small air compressor. Don't forget to use M7 for Air Assist in LightBurn. 
[image]

You can configure it as GRBL.

manuelchaves, 2023-02-25:
> You can use G and SCL. I use an SSR with low voltage trigger (SCL delivers only 3.3v). In my case a SSR-40DA with an small air compressor. Don't forget to use M7 for Air Assist in LightBurn. [image]
> 
> You can configure it as GRBL.

Tks

closed 2023-02-27 by abinoer