makerbase-mks/MKS-DLC32 #347: LaserBox ESPP32 v1.0 (laser does not fire)

MJ666, 2023-02-26:
I installed https://github.com/makerbase-mks/MKS-DLC32/blob/main/firmware/TS35/CNC/Normal/Board_V1.0/V2.0.8_H35_20220105_N_CNC.bin to the LaserBox ESP32 V1.0 controller. This is a MKS DLC32 V1.x clone board used by Atomstack for there more actual lasers. All functions are working except the laser does not fire. I than tried to build a Firmware from the released source code. I get this to work and using the laser. https://github.com/MJ666/MKS-DLC32-FIRMWARE/tree/LaserBoxESP32 Unfortunately the source code looks to be older than what was released as binary firmware. The display does not work as well as the prebuild version. I wold be nice if you can release the source from recent builds

abinoer, 2023-02-27:
Our firmware may not match the motherboard you are using (LaserBox ESPP32 v1.0) and the laser does not fire

closed 2023-02-27 by abinoer

MJ666, 2023-02-28:
Finally firmware [https://github.com/.../Board_V1.0/V2.0.8_H35_20220105_N.bin](https://github.com/makerbase-mks/MKS-DLC32/blob/main/firmware/TS35/Laser/Normal/Board_V1.0/V2.0.8_H35_20220105_N.bin) is working on the LaserBox controller including the display. I think [https://github.com/.../V2.20_H35_2022_1101_N_ZX_002.bin](https://github.com/makerbase-mks/MKS-DLC32/blob/main/firmware/TS35/Laser/Normal/Board_V1.0/V2.20_H35_2022_1101_N_ZX_002.bin) firmware is broken and will not work on an MKS DLC V1.1 for laser use. Would be really good if you can release the latest source code so the community can help to improve. After playing with this source code for a couple of days i gave up since there are to many problems in there wich are already be solved in the binaries you published.