makerbase-mks/MKS-DLC32 #348: How to start the homing in the display

Badboy1986, 2023-02-27:
I use the MKS-DLC32 with TS35 display on a CNC 3040 with the firmware V2.1.0_H35_20220526_02_N_CNC.bin. When using the machine without limit switches, I was able to operate all necessary functions on the display without a PC. Now, unfortunately, after installing and activating the limit switches, I cannot start the homing via the display and need the PC or my smartphone with the Web UI to start the $H command manually. 

Is there a way to do this comfortably via the display in order to carry out the complete offline operation without a PC? Does anyone here have a modified firmware with an adapted display menu?

milpaisajes, 2023-03-02:
Send $$ and check the parameters $21 and $22 you have to activate $21=1 and $22=1

Badboy1986, 2023-03-05:
I only have $20=1 and $22=1, $21=0 because I have only installed limit switches on the zero point side. I have no limit switches on the maximum position. For this reason I thought it was wrong to activate the hard limits.