makerbase-mks/MKS-DLC32 #351: Relay connection

TomasKliner, 2023-03-07:
Hello, i was wondering how would i go about connecting **Relay** to the board in a way that would Turn on at the same time as laser engraver. 

In the past i tried connecting standard relay to the ttl port but since its PWM signal i tried to switch it for 12v PWM RC controlled relay, but that seems to be having issues also. I assume the TTL port is not suitable/compatible. 

Is there any other way that i could connect to the board a standard **5v relay**, that would turn on at the same time as the laser ?
_(Trying to create simple plasma cutter, using lasergrbl)_

AndWalu, 2023-03-07:
It will be easier to attach 12V relay:
[image]

5V relay:
[image]

These are just theoretical schematic, I didn't do them.

The relay should remain on for 10-20 ms after the pulse disappears.

dart1280, 2023-05-11:
Why not use the cooling output that already exists?. arrange via post processor or gcode generator to send a M7 before each M3 and an M9 to switch it off after each M5 in your gcode. Fit a relay board like [https://www.aliexpress.com/item/1005004306855775.html](url) to SCL as per the instruction manual?

[image]