makerbase-mks/MKS-DLC32 #352: Wrong firmware for MKS DLC32 V1.0 board

MJ666, 2023-03-08:
The latest released laser Firmware for V1.1 boards is wrong. This firmware is working for an V2.x board. The laser does not fire on an v1.1 board but on an 2.1 board. 

https://github.com/makerbase-mks/MKS-DLC32/blob/main/firmware/TS35/Laser/Normal/Board_V1.0/V2.20_H35_2022_1101_N_ZX_002.bin

It would be nice if there are also more recent releases for v1.1 boards are provided. If the latest source code will be released the community can help to improve the code. The actual released code is outdated and much older then what is released as binary firmware.

msy021, 2023-03-10:
This has been mentioned in couple of issues until now but they were closed by @abinoer without any real solution.
#309 #347 
Maybe with this one we will have better luck.

An alternative would be [FluidNC](https://github.com/bdring/FluidNC), but for now it does not support the screen, and from issue [#759](https://github.com/bdring/FluidNC/issues/759) it looks like they are not willing to support the screen too soon.

MJ666, 2023-03-10:
Yes im aware of this issues and feature requests. I also played with FluidNC already. Im actually running:  https://github.com/makerbase-mks/MKS-DLC32/blob/main/firmware/TS35/Laser/Normal/Board_V1.0/V2.0.8_H35_20220105_N.bin at my Atomstack X20 mainly for the reason to be able to control air assist and switch the fan of the FB2 enclosure via M7/M8/M9 commands in Lightburn. Atomstack is using an MKS DLC32 V1.1 clone controller (LaserBox ESC32 V1.0). The Atomstack firmware does not support this feature and they locks to have no interest to add this into there firmware. I'm not sure when the Atomstack firmware was forked from the MKS source and which version is more up to date? Unfortunately the released MKS source code is quite outdated and has some problems. I already played with an custom build from this source. It looks to be MKS is not releasing the latest source code. The community could really help to improve this.