makerbase-mks/MKS-DLC32 #354: TS35-R v2.0 firmware missing control Z axis buttons

onehans1, 2023-03-10:
i really miss controlling Z axis via screen. could someone please add buttons to laser mode?
also simple "$HZ" button macro would of be nice to have

onehans1, 2023-03-10:
latest v2.30 (8M.H35.20221223)

cheers

onehans1, 2023-05-03:
Tbh i couldn't be arsed to upload older firmware, I'll just wait for new one. Hopefully this time they won't forget about Z axis and few macro buttons

diy-laser, 2023-06-08:
The same here !!
Manual via display no Z control.
Shame....
Always worked with TS24 now switched to TS35.
Then I thought.... flash new firmware.... WRONG
I have now gone back a few steps with a working Z control.
V2.10_H35_2022_0621_N_ZX_001 this works.

Tashadra, 2023-12-30:
@makerbase-mks please bring Z control back, removing it seems like a step backwards.