makerbase-mks/MKS-DLC32 #364: Touch screen scrambled

Chriswoodywoods, 2023-04-06:
Hi. I have an issue with a new board and screen. The board powers up fine and the screen will power on. But, the screen is "scrambled" as if there is a wiring or firmware problem. The cables are connected properly with exp1 to exp1 and exp2 to exp2. There are some letters that create a word such as "disconnected" but is backwards as "detcennoncsid" but in French I believe. 
[image]

manuelchaves, 2023-04-06:
Upgrade your firmware to correct display version. Your display is TSR-24R and firmware in MKS-DLC is  to TSR-35R version

Chriswoodywoods, 2023-04-06:
I tried doing a firmware install and the port on the board must have a
problem. It will not connect to my laptop. I have the drivers installed on
my laptop and use it almost daily to connect a raspberry pi. I tried
different settings in the flash program from the github page. The github
page is also out of date to what to the program is so maybe there is
something there about that. Plus why would they ship a board with the
incorrect firmware? Bad quality control? Human error?

manuelchaves, 2023-04-07:
Understand, I don't know why they still do this with the tS24 display, it's very basic and subtracts relevant information. This MKS-DLC32 project was a good idea, but let it die. But it solves my problems, but...
I think it's been abandoned and I can't find a low-cost alternative.
As for updating the Firmware, I know it's illogical, but try to do it in Windows 7 or 10, without antivirus and the like. The flasher is very rough