makerbase-mks/MKS-DLC32 #373: MKS LTS V1.1

willbonholiday, 2023-04-24:
Evening. I have recently purchased a two trees TT-55 laser engraver. Board says its an MKS LTS V1.1.
Firmware seems to be dead and the supplied USB stick didnt have any firmware included.

Two trees website is horrendous with no help and their downloads area is just worse.

Cant seem to find anything online regarding the MKS LTS 1.1 and cant even seem to find any info on here.

Can anyone point me in the right direction of the correct firmware to use please?

acguardia, 2023-04-26:
Download the latest or second to latest firmware update for version V1.0 of the DLC32 board and 3.5" LCD.
Do the upload of this eeprom file throut MKS Laser Tool -> Parameter Configuration
[MKSConfig_TTS_55.zip](https://github.com/makerbase-mks/MKS-DLC32/files/11327687/MKSConfig_TTS_55.zip)
[image]

salacpavel, 2023-04-27:
Following, as I have the same issue. Could anyone please point me to firmware binary?

s752313, 2023-05-03:
Hi! has anyone found the firmware?

eastwooddotai, 2023-05-04:
[LKS_TT_2022092004.zip](https://github.com/makerbase-mks/MKS-DLC32/files/11395250/LKS_TT_2022092004.zip)

@s752313 @salacpavel after some email tennis TT managed to send me the attached firmware

DougJoseph, 2023-07-01:
I have the same TwoTrees laser rig (TTS-55) and the MKS touch screen (MKS TS35-R v2.0) had I flashed a DLC32 firmware version 2.3, as I was hoping for landscape mode, a more colorful interface, and some issues to be fixed, but the newer firmware broke the rig. I got an instant, constant beep with a false warning about the laser being on fire. The interface seemed unresponsive. So then I used the TT firmware shared above by  [eastwooddotai](https://github.com/eastwooddotai) commented [on May 4](https://github.com/makerbase-mks/MKS-DLC32/issues/373#issuecomment-1534276217). That got me back to the same vertical (portrait) interface, with the mostly B&W look, and some issues including seriously diminished power as compared to before I installed the touch screen. I tried changing the GRBL setting for max power from 1000 to 255 (and changing LightBurn device setting to match), but it did not help, so I put both back to 1000. Switching on "constant power" in LightBurn layers option increases power some, but I'm still disappointed in the lower power from a 5.5W laser.

DougJoseph, 2023-07-01:
UPDATE: Whenever I tried the latest version of the DLC32 firmware, for "Board_v1.0", the laser would not fire. However, I then backed the firmware version down to 2.0.8_H35, dated Jan 5, 2022, and it gets me a working laser, and in landscape mode with a colorful interface. 

Here’s the firmware that works:
V2.0.8_H35_20220105_N.bin

https://github.com/makerbase-mks/MKS-DLC32/blob/main/firmware/TS35/Laser/Normal/Board_V1.0/V2.0.8_H35_20220105_N.bin

andreasr81, 2023-07-14:
> UPDATE: Whenever I tried the latest version of the DLC32 firmware, for "Board_v1.0", the laser would not fire. However, I then backed the firmware version down to 2.0.8_H35, dated Jan 5, 2022, and [...]

Thanks, i have the same issue.

MJ666, 2023-09-07:
> UPDATE: Whenever I tried the latest version of the DLC32 firmware, for "Board_v1.0", the laser would not fire. However, I then backed the firmware version down to 2.0.8_H35, dated Jan 5, 2022, and [...]

The latest firmware version in the V1 section is for an V2 board. I reported about this already but no reaction.

kevinsikes, 2024-01-01:
I was successful in compiling the firmware for the MKS LTS v1.1 board by commenting out the two `#define USE_BOARD_V2_0` lines in the header files.  Then the laser fires correctly.

DougJoseph, 2024-01-01:
👍

MJ666, 2024-01-01:
Unfortunately the source code on git is older than the firmware delivered as binaries. There are still issues with the touch screen and more. I found the V2.0.8_H35_20220105_N.bin firmware is the better option for an V1 board.

kevinsikes, 2024-01-01:
I tried V2.0.8_H35_20220105_N.bin but it got into a reboot loop.  My rig doesn't use a touchscreen -- are there other functional issues to be aware of when compiling from the sources they provided?

MJ666, 2024-01-01:
Did you use this version? 

https://github.com/makerbase-mks/MKS-DLC32/blob/main/firmware/TS35/Laser/Normal/Board_V1.0/V2.0.8_H35_20220105_N.bin 

If you get a bootloop with this one it is strange.

fernando199601, 2024-02-25:
I have a problem with my CNC laser with MKS LTS V1.1 board without display. While in use it was paralyzed I think the firmware is fine but I can't load it successfully.

answer2420, 2024-07-03:
alguien sabe como agregar el firmware ?

jeanfrodrigues, 2024-09-18:
someone know how to solve de problem with WiFi disconect after reboot de board?