makerbase-mks/MKS-DLC32 #374: MKS DLC32 V2.1 and MKS TMC2009 V2.0 in UART Mode

Jet-A1, 2023-04-29:
Hello,
i bought a MKS DLC32 V2.1 Set with MKS TMC 2009 Driver and TF-35 Display from Aliexpress. When i buy a Set with Board and Driver i want to benefit from the Features the TMC 2009 comes with. The Videos and the Dokumentation are very short on Information.

Can you please explain how to configure the Board in UART Mode that i can benefit from the Features the TMC 2009 comes with ?

MJ666, 2023-04-30:
With the DLC32 the uart mode of the TMS2009 can not be used. It is not supported by the hardware design.

real-artswan, 2025-01-19:
I found that such a modification can be made with MKS Tynibee here http://wiki.fluidnc.com/en/hardware/3rd-party/MKS_TinyBee. And also I found that UART1 pins on the ESP are not connected 

[image]

May be it is possible to connect DLC32 in a similar way. 
I'd like to try but is it worth to  tinker with soldering? May be I'm missing and it will not work, idk.