makerbase-mks/MKS-DLC32 #375: Screen TS35R

GabrielxSilva, 2023-05-02:
Olá, como faço para mudar a logo inicial do meu display laser, TS35R v2.0? Da pra cololocar video ou apenas imagem ?

[image]

https://user-images.githubusercontent.com/132404378/235789250-489819f2-e420-441d-8ffc-567bb47d34e9.mp4

manuelchaves, 2023-05-02:
Gabriel, boa noite, veja este link

https://github.com/makerbase-mks/MKS-DLC32/issues/127

GabrielxSilva, 2023-05-03:
> Gabriel, boa noite, veja este link do Inssues #127 . Só imagem e mem simples, vídeo nem pensar
> 
> #127

voce pode fazer a imagem para mim ?
[image]