makerbase-mks/MKS-DLC32 #377: newly getting into this. First startup question for display/laser/cnc firmware

klingklang01, 2023-05-06:
Hello,  I am building myself a desktop CNC,  That was the easy part.  Now I need to have everything works and talk to each other.  Just started the DLC32.  Installed the latest firmware for my version and as I want it for CNC and have  a TS35-R, I've downloaded the V2.1.0_H35_20220526_02_N_CNC.bin file.  Surprise!   screen have power but nothing to show.  After a few test, I came to the conclusion that everything work with the firmware for the LASER but don't with the CNC firmware.  One question:  WHY?

klingklang01, 2023-05-09:
I got it work but with the old V2.0.8_H35_20220105_N_CNC firmware .  how come??????

JSF9026, 2024-11-23:
Hi,
can you please share the old cnc firmware as i can not find it to download it ?
Thanks