makerbase-mks/MKS-DLC32 #379: Error21

small66666, 2023-05-09:
MKS DLC32 2.1

Описание машины
3-осевой лазерный станок ЧПУ с драйверами TMC2209

Файл конфигурации
$0 = 7 (время импульса шага)
$1= 255 (задержка холостого хода шага)
$2= 0 (Инвертирование импульса шага)
$ 3 = 0 (Инвертирование направления шага)
$ 4 = 0 (Инвертирование вывода включения шага шага)
$ 5 = 1 (Инвертирование выводов ограничения)
$ 6 = 0 (Инвертирование вывода датчика)
$ 10 = 1 (Параметры отчета о состоянии)
$ 11 = 0,010 (Отклонение соединения)
$ 12 = 0,002 (Допуск дуги)
$ 13 = 0 (Отчет в дюймах) 
$ 20 = 1 (включены мягкие ограничения)
$21 = 1 (включены жесткие ограничения)
$ 22 = 1 (включен цикл самонаведения)
$ 23 = 3 (Инвертировать направление самонаведения)
$ 24 = 300 000 (Скорость подачи сигнала определения местоположения самонаведения)
$ 25 = 1000 000 (скорость поиска самонаведения)
$ 26 = 25 000 (задержка отключения переключателя самонаведения)
$ 27 = 2000 (расстояние срабатывания переключателя самонаведения)
$ 28 = 1000 000 (частота вращения шпинделя)
$ 30 = 1000 000 (максимальная скорость вращения шпинделя)
$ 31 = 0,000 (минимальная скорость вращения шпинделя)
$ 100  = 1600 000 (разрешение перемещения по оси X)
$ 101  = 1600 000 (разрешение перемещения по оси Y)
$ 102  = 80 000 (разрешение перемещения по оси Z)
$ 110  = 2100 000 (максимальная скорость перемещения по оси X)
$ 111  = 2100 000 (максимальная скорость перемещения по оси Y)
$ 112  = 6000.000 (максимальная скорость перемещения по оси Z)
$ 120  = 100 000 (ускорение по оси X)
$ 121 = 100.000 (ускорение по оси Y)
$ 122 = 500.000 (ускорение по оси Z)
$ 130 = 410.000 (максимальный ход по оси X)
$ 131 = 282.000 (максимальный ход по оси Y)
$ 132 = 80.000 (максимальный ход по оси Z)

Startup Messages
Ожидание соединения...
ok
[PRODUCER:MKS DLC32]
[V2.30(8M.H35.20221223)]
[VER:1.1h.2022122301:]
[OPT:VMPH,63,500]
Target buffer size found
[MSG:Using machine:MKS DLC32]
[MSG:Mode=AP:SSDI=ESPWIFI_16431:IP=192.168.4.1:MAC=C4-DE-E2-37-2F-41]
ok
User Interface Software
LightBurn

Что произошло?
Если использовать Wi-Fi соединение и в настройках программы ( Режим передачи - Буферизованный ) то
Layer 1
error:21
В блоке обнаружено более одной команды g-кода из одной модальной группы.
На или около линии 28:
Задание остановлено
Поток завершен в 0:00
При подключении через USB то нет этой ошибки.

Babaika1987, 2023-05-12:
Are there any living ones?

small66666, 2023-05-12:
Is anyone alive?

Babaika1987, 2023-07-30:
Дайте решение.

Andron66, 2024-03-28:
Есть решение этой проблемы?

BaBaika666, 2024-03-28:
Нет. Перешёл на usb соединение

Andron66, 2024-03-28:
Жаль

baunt78, 2025-04-14:
Спасибо автору что написал свою проблему подробно, имею Makerbase MKS LS ESP32 PRO столкнулся с той же проблемой... ток у меня гравер и он останавливался... Автор написал "Режим передачи - Буферизованный" и тут в голове возник вопрос буферизация значит гдето должно записываться и держаться в памяти... моя плата я так понимаю не умеет держать буферизацию... ну и че делать поменял ее на синхронизированный режим и уаля работает не останавливается... 

[image]

baunt78, 2025-04-16:
Кому помогло?