makerbase-mks/MKS-DLC32 #380: Homing cycle problems

Codenewbie420, 2023-05-13:
So my homing cycle begins normally on my laser engraver, both x and y end switches are hit, both axis move back a little but then they hit the end switches again and instead of stopping, the motor keeps spinning even when the end switch is applied, the motor spins for a couple of seconds and then stops, making a horrible noise and skipping the belt. Has anyone had any similar issues or has any idea how to fix it?

SirShrotALot, 2023-06-21:
Have you checked if your axis move back wide enough so the switch is not triggered anymore before it hit the switch again? I had this issue today.

Codenewbie420, 2023-06-24:
i had checked, it does move away far enough, thanks for the suggestion :)

sharknetwork24, 2023-07-12:
I've h ad the same problem.
Since I replaced the board the problem has disappeared.
Please double check the links for information.