makerbase-mks/MKS-DLC32 #381: Origin position

rominetb44, 2023-05-21:
Hi,

With my K40, the origin of the homing is top left, so the positive direction is to the right and to the bottom.
So, in the moving page of the TS35, the arrow are inverted (positives values are for the top arrow and left arrows).
Could you allow to configure it or rebuild the V2.10_H35_2022_0621_N_ZX_001.bin file for inverted machines ?

Thanks.

SirShrotALot, 2023-06-21:
I'm facing the same problem. But I also have the problem that the arrow on the display and on the website are doing different movements!
Up arrow on the display move the Y-axis **down**. Up arrow on the website move the Y-axis **up**!
Left arrow move the display move the X-axis **right** and the left arrow on the website is moving the X-axis also the **right**.

Abdullah199220, 2024-03-13:
have a nice day . I have that same problem . limit switches are at the top right. When I move the target position and origin to the top right, the ts35r buttons move correctly, but the y axes move in the opposite direction in the lightburn movement commands. (I was able to achieve this with $3) I would be happy if anyone can help me find a solution.

rominetb44, 2024-03-14:
Hi,
The issue of the arrow is fixed in the V2.30_H35_2022_1223_N_ZX_001.bin firmware.However, in lightburn you have to change the origin of your machine.

Thanks