makerbase-mks/MKS-DLC32 #383: DLC32 v2 Laser is always on

sharknetwork24, 2023-06-07:
Everything works perfectly except that the laser is always on and to turn it off I have to disconnect the general power supply.
I would like to know if the problem has been solved, or if it is a defect of the Board DLC32 v2 that does not work.
Can it be solved or should the BOARD be replaced?
Thanks for the replies.

sharknetwork24, 2023-06-07:
I connected everything correctly.
I attach image of the laser module where the DLC32 v2 board is connected to the laser module

[image]

[image]

[image]

sharknetwork24, 2023-06-08:
Is it possible to have a wiring diagram to understand where the problem lies?
And the mosfet to replace Q1 (WSF40N10)?
On J18 I always have a voltage of 12.35V

Ho trovato questo ma non so se è corretto.
[image]

3Dangel, 2023-06-10:
don't use the TTL
USE THE Spindle.
if you found the schematic of v2.x then chech with the schematic of V1.x and you will see the differences between the two. 
V 1.x will work. V2.x won't.

sharknetwork24, 2023-06-10:
> don't use the TTL
> USE THE Spindle.
> if you found the schematic of v2.x then chech with the schematic of V1.x and you will see the differences between the two. 
> V 1.x will work. V2.x won't.
> 

Is the spindle output ok to control the laser?  
So you're telling me that the 3-pin TTL output doesn't work in version 2.1 boards?

sharknetwork24, 2023-06-10:
If anyone knows how to make the links please use the picture and re-post with the correct links. 
Other people may benefit. 
Thank you

[image]

sharknetwork24, 2023-06-13:
May be of help to others who have the same Laser.

[image]

[image]

closed 2023-06-22 by sharknetwork24