makerbase-mks/MKS-DLC32 #390: Customize start screen MKS DLC32

sharknetwork24, 2023-06-22:
Create your personal image strictly with these dimensions:
DIMENSIONS 480 x 320

WIDTH 480 Pixels
HEIGHT 320 Pixels
DEPTH IN BIT 24

Use Photoshop, Corel Draw, Inkscape etc.

sharknetwork24, 2023-06-22:
STEP 3

Install and open MKS Laser Tool V2.0.4 which you can download here:

https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware/tool

Connect your MKS DLC32 board
Open MKS Laser Tool V2.0.4 and click WiFi Configuration Tool

[image]

sharknetwork24, 2023-06-22:
STEP 4

Select the COM port and Connect
Enter your WiFi name and Password.

[image]

[image]

Copy the IP address and enter it on your browser.

sharknetwork24, 2023-06-22:
STEP 5

Opened the browser and opened your IP.
Click on About

[image]

sharknetwork24, 2023-06-22:
STEP 6

Click on Update WEBUI

[image]

Now click on:
index.html.gz
mks_logo.bin
Save them to your computer for safety so you can restore them if something goes wrong.

[image]

After saving the 2 files just delete mks_logo.bin

[image]

sharknetwork24, 2023-06-22:
STEP 7

You will find yourself in this situation after deleting the mks_logo.bin file

Now click on Select files and load the .bin image you created earlier.

Loading will be immediate wait 15 sec. about and you will see the 2 files again.

Now disconnect everything and restart the board and as if by magic you will see the screen splash with your custom image.

[image]

After uploading your .bin file you will have this situation.

[image]

K4o0s2, 2023-06-22:
I must tell you that YOU ARE A FUCKING MACHINE!!!!

Fantastic tutorial well worth a video.
I want to point out that I have skipped the MKSLaserTool. I configured my home wifi on the card so when it connects I just have to look at the IP and it works. The rest was easy with your tutorial.
A million thanks for the work and for making life easier for everyone.
Now if we have the latest firmware with the custom screen. As they say in my country... for sample, a button. I leave image of my current screen.

Thank you very much once again! @sharknetwork24
[image]
(url)

sharknetwork24, 2023-06-22:
> Devo dirti che SEI UNA MACCHINA DEL CAZZO!!!!
> 
> Tutorial fantastico che merita un video. Voglio sottolineare che ho saltato MKSLaserTool. Ho configurato il mio wifi di casa sulla scheda quindi quando si connette devo solo guardare l'IP e funziona. Il resto è stato facile con il tuo tutorial. Un milione di grazie per il lavoro e per aver reso la vita più facile a tutti. Ora se abbiamo l'ultimo firmware con la schermata personalizzata. Come si dice nel mio paese... per esempio, un bottone. Lascio l'immagine del mio schermo attuale.
> 
> Grazie mille ancora una volta!@sharknetwork24 [image] (URL)

I'm glad you managed with my help.
It's nice to help those in need.
Congratulations for the image.

K4o0s2, 2023-06-23:
I would like, of course with your permission, to make a small video with your tutorial for youtube of course with your nick and links that are convenient for you. What do you think? I am sure that in my country I am not the only one with this problem and you have masterfully solved it

sharknetwork24, 2023-06-23:
I give you permission to make the video.  Then put the link to the video.

closed 2023-07-12 by sharknetwork24

burak1000, 2023-08-01:
this only works with 2.3 version. but CNC folder doesnt have 2.3 version bin file. pls help. thanks

burak1000, 2023-08-01:
also i tried on 2.3 laser firmware it's not working too. i dont understand why?