makerbase-mks/MKS-DLC32 #395: Laser won't fire + red DEL on MKS DLC32

SLB0810, 2023-07-03:
Hi !
First message here; so hello to the community!
I've recently purchased a MKS DLC32 board for my 3018 laser machine.
Everything seems to work properly (movement on all 3 axis, switches for X and Y stops, link with PC / Lightburn). However, there is a red light ON all the time and the laser won't turn of when I hit "fire" or when I launch something to be engraved. The laser moves properly, but there is no laser light.
The $30 and 32 are set at the correct values (resp. 10000 and 1).
When I use the laser with an the stock card provided with the 3018 laser, it works (but I changed as I had no pin for the end-stops on this one).
Can someone tell me what I've done wrong and how to correct it ?
Best regards,
Sébastien
[image]

SLB0810, 2023-07-05:
Quick additionnal comment : the red light remains ON even if everything apart the power is disconnected (no stepper, no motors, no laser, no endstop switch, no USB connection, etc.). The providers states that the red lights means that an order is given to the laser but I don't understad how it could be possible as everything is disconnected...
Any help is much welcome, I have some project that I had to stop because of this :-/

maadmackz, 2023-07-06:
put laser fire on the spindle -, use this as your pwm.

Delson123, 2023-07-11:
Could you be kind enough the tell me what type of terminals you are using ???

SLB0810, 2023-07-11:
Hi ! Thank you all for your feedbacks !!

@sharknetwork24 : I've flashed the latest version I found under this github folder : https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware/TS35/Laser/Normal/Board_V2.0

Here are the photos of the MKS DLC32 with the laser connected and how I connected it to the laser.
After some discussions with the motherboard provider, they said that the red light shows that a signal is sent to the laser, but :
- the red light appears even if the motherboard is not connected to the computer
- when I plug the laser, it won't fire.

@Delson123 : the basis is a cheap 3018 laser machine. I use Lightburn on a windows computer (which worked perfectly with the stock motherboard. But I had no pin to connect the end-stop switched, so I upgraded it to the MKS DLC32). the drivers are installed on the computer (I don't remember the name, CH340 or something like that).

@maadmackz : what would be the wiring to do so ? Do I let everything connected to the laser output, except the PWM that I connect to the spindle output ?

Once again, thank you all for your support !

[image]
[image]

sharknetwork24, 2023-07-11:
See to take a cue of how the LASER should be connected.

[image]

[image]

SLB0810, 2023-07-29:
Hi again,
Thank you for helping me out, and sorry for the late reply, I was on leave with no possibility to take the picture of the laser.
So here it is ! 

[image]

@sharknetwork24 : thank you for the illustration, but I'm not sure it applies to my configuration as I don't have this card between the MKS DLC32 and the laser. All I have are the PWM / GND / VCC input on the laser, and the S / TTL / VIN outputs on the MKS DLC32. How does it impact the wiring you linked ?

I'll try to change the software with an older version to see if it helps.

Thank you again,
Sébastien

Benglin, 2024-07-27:
2 days to your post's anniversary, @SLB0810, I hope you have found a solution to your problem 🙂 You would not have the "card" between MKS DLC32 and the laser because yours is probably built into the laser module (right below that fan?). For that your connection from `S/TTL/V` on the MKS DLC32 should go directly to the laser module (see the first image in [my question post](https://github.com/makerbase-mks/MKS-DLC32/issues/493)).

⚠️ Since your laser module is 12V, you can only power your MKS DLC32 with 12V, otherwise the `V` in `S/TTL/V` will be too high for your laser module, that'll burn it.

Benglin, 2024-07-27:
2 days to your post's anniversary, @SLB0810, I hope you have found a solution to your problem 🙂 You would not have the "card" between MKS DLC32 and the laser (like what @sharknetwork24 showed) because yours is probably built into the laser module (right below that fan?). For that your connection from `S/TTL/V` on the MKS DLC32 should go directly to the laser module (see the first image in [my question post](https://github.com/makerbase-mks/MKS-DLC32/issues/493)).

⚠️ Since your laser module is 12V, you can only power your MKS DLC32 with 12V, otherwise the `V` in `S/TTL/V` will be too high for your laser module, that'll burn it.

Benglin, 2024-07-30:
Hi @SLB0810, just so you know, I have resolved my [wiring issues](https://github.com/makerbase-mks/MKS-DLC32/issues/493) in my scenario, just in case it helps you too.