makerbase-mks/MKS-DLC32 #398: problema ao cortas peças na co2 algumas linhas aplaca nao reconhece e não corta

orlandorezende, 2023-07-12:
estou com problema serio uso uma placa mks 32 e o light burn e quando coloco para cortar determinada peça com vários desenhos muitas linhas ficam sem cortar essa placa não recomendo a ninguém mais me arrependi de ter comprada ela não tem suporte técnico nenhum

manuelchaves, 2023-07-12:
Orlando, bom dia, duas dicas: 
No LightBurn dminua a velocidade de aceleração de corte e a tal de velocidade rápida (que até hj não consegui me entender com ela) teste as combinacoes de velocidade minima de canto e precisao de canto, acho que não tem uma formula unica mas tive otimas respostas por ai
Na DLC32 eu tive muito probelma com os TMC2226 e 2225, so reolvi quando comprei os tmc2209 da propria makerbase

Boa sorte

orlandorezende, 2023-07-12:
Olá meu amigo Cristóvão o grande problema meu aqui e se eu coloco um
desenho com muito detalhes tem várias linhas que não corta entende já
diminui velocidade já ajustei minha máquina já fiz de tudo oque você
imagina e também já corrigi vetores abertos confiro em 3 programas pra ter
a certeza que eles não está aberto ou seja estou insatisfeito com essa
placa ela não é capaz de fazer a leitura correta e como ela reconhece uma
linha e diminui a potência do laser entende me ajuda aí se possível

orlandorezende, 2023-07-12:
> linha e diminui a potência do laser entende me ajuda aí se possível detalhe
> uso driver externo já para não forçar a placa em relação à alimentação

sharknetwork24, 2023-07-12:
The problems you described do not happen to me and I believe also to the majority of users who use DLC32.
I confirm that the best drivers to use are the TMC2209.
If the board does not have the correct settings with lightburn it will never work properly.
I use it with a diode laser and I don't find any problems.
Have you taken into consideration that the board in your possession works perfectly?
Maybe he has some problem?
In the past I have had a defective board and replacing it solved all the problems.
I advise you to buy directly from MKS so you will also have assistance.

https://it.aliexpress.com/item/1005003183498253.html?spm=a2g0o.order_list.order_list_main.15.60773696rKQQ2Z&gatewayAdapt=glo2ita