makerbase-mks/MKS-DLC32 #400: Air Assist PIN for DLC32 V1.0 Issue

zakkaon, 2023-07-24:
Hi friend, I bought sculpfun s9 laser engraver, where the board using mks. 

[image]

I need help. How do I get signal from lightburn software. To control air assist. Which pins can I use with this version of the board. ?? Thank you very much.

MJ666, 2023-08-14:
If this is running MKS firmware you can try to locate the IO04 pin. If it is not exposed you can solder a wire directly to the related ESP pin.

Wiki-the-Tinkerer, 2023-09-03:
maybe check https://github.com/makerbase-mks/MKS-DLC32/wiki/Hardware-Wiring

ElectricBluefishProductionsInc, 2023-11-21:
SCL pin is high by default on my board. Obviously when I connect it to my relay module it sets the relay ON which is not what anyone would expect. Can the default state of the SCL pin be set to low so that we can turn it on and off with the G-Code command when required/cutting? My relay module 

Also, Is it possible to use the I2C header and protocol to address a I2C relay? 
[image]

Wiki-the-Tinkerer, 2023-11-21:
> There are relay modules available wich have an opening and a closing contact. This can solve your problem.
> 
The relays in the drawing looks to have both NC Normally closed and NOpen. So should be no prob, as long as you get a signal to them, either way.

ElectricBluefishProductionsInc, 2023-11-22:
I will give that a try... I did not consider reversing the relay, was focused on the board.

Akenaton329, 2025-03-02:
Hi guys, I made this connection on my MKS-DLC32 2.0 board with the latest firmware. I’m trying to connect the air assist, but I’ve noticed that from the SCL pin, when I connect it to GND, there is no voltage when I activate the program with M8. Could you help me

[image]