makerbase-mks/MKS-DLC32 #405: need help for wiring DY13 co2 laser power supply to mks dlc32

Enzo-Enz, 2023-08-25:
Hello my friends
Recently I bought a co2 laser and I couldn't run the laser 
could you please tell me which pins of mks dlc32 must to be connect to the laser power supply

here is the power supply pins

[image]

Enzo-Enz, 2023-08-28:
> i did start the laser by connecting s>in . ttl>TL . vin>5v . G&WP>g(on the board) but it fire only with 50 percent power or more i don't know how to run laser under 50 percent and i'm trying to find a way to do that

Thank you bro I'll test that

for your issue check your ttl frequency

and where are you from? I think we are compatriot

AmirBAV, 2023-08-30:
You're welcome. I made some changes and now the laser tube works under 50 percent but still has some problems. I'm from Iran

AmirBAV, 2023-08-30:
[image]
Here is my diagram that works fine with 10 percent power to 100 but the problem is the laser LED is always on even when power is 0 it works ok but I guess it will affect tube life eventually

Enzo-Enz, 2023-09-01:
> You're welcome. I made some changes and now the laser tube works under 50 percent but still has some problems. I'm from Iran

I'm from Iran too bro
this is my telegram ID"ehn963"please send me a massage to fix the issue together I bought this laser too

Wiki-the-Tinkerer, 2023-09-08:
Just been measuring TTL output from the 3pin output (mainly for laser, but also used by external VDF:s, as it's supposed to be 0-5v). Discoveries:
1) The pins seem to be marked "wrong"(!).  The pin order is Signal-Ground-V(12-24v). On the board it looks like the middle pin is signal, as it says S-TTL-Vin. At least an easy mistake to make.
2) There seems to be a SERIOUS calculation error in the firmware, which is downright dangerous!!   
I have one machine that has a max of 24000 rpm. So in the grbl settings $30=24000. Now the expectation is that 24000rpm gives out 5v, and 0rpm 0v.

When testing, 0 gives out 0. good. BUT 1000rpm gives out 2.05v, 2000 rpm 4.11v, 3000rpm 1.23v.. and 24000 rpm gives out 0.

So there probably is some "base" assumption rpm value for 100% - and not saying this in big letters at the front of all the channels is potentially lethally dangerous.

So measure / test what values in the settings seem to give out 100%, and doublecheck values in between, to be sure. This thing could burn houses or take a mans arm off.

Benglin, 2024-07-27:
I'm glad I saw your comment, @Wiki-the-Tinkerer! 😄 I have somewhat same (and somewhat different) issue with the S/TTL value reading (for me the PWM ranges in `0 ~ 12V`). I [posted details of it](https://github.com/makerbase-mks/MKS-DLC32/issues/493) yesterday, hopefully someone like yourself can help me out with fixing this PWM range. Perhaps I should flash the firmware, hmmm... 🤔

Benglin, 2024-07-30:
Hi @Wiki-the-Tinkerer, just so you know, I have resolved my [wiring issues](https://github.com/makerbase-mks/MKS-DLC32/issues/493) in my scenario, just in case it helps you too.