makerbase-mks/MKS-DLC32 #407: DLC32 start and stop movements losing position

tjivanhoff, 2023-08-26:
Premise: with my old genmitsu everything was ok.

A simple square on lightburn. The machine engraves well the first and the second edges. Starts the third one. Somewhere it suddenly stops for a fraction of second and then restart, and again, and again. At the end, the third edge is shorter. Starts the fourth one. Stops and starts, stops and starts. The fourth edge too is shorter then the third one.

At the moment the microstep configuration is ON-ON-ON, but the problem is the same with every setting.

You can imagine what happens with complex designs...

Where is the problem?
 

Configuration: DLC32+TS35+DRV8825
Firmware:
[PRODUCER:MKS DLC32]
[V2.30(8M.H35.20221223)]
[VER:1.1h.2022122301:]
[OPT:VMPH,63,500]
Target buffer size found
[MSG:Using machine:MKS DLC32]
[MSG:No Wifi]

GRBL:

$0=10 ; somewhere I found that could help but... nope.
$1=25
$2=0
$3=4
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=300.000
$25=1000.000
$26=250.000
$27=1.000
$28=1000.000
$30=1000.000
$31=0.000
$32=1
$50=1
$38=1
$40=1
$100=1600.000
$101=1600.000
$102=1600.000
$103=100.000
$104=100.000
$105=100.000
$110=2000.000
$111=2500.000
$112=500.000
$113=1000.000
$114=1000.000
$115=1000.000
$120=25.000
$121=25.000
$122=25.000
$123=200.000
$124=200.000
$125=200.000
$130=282.000
$131=335.000
$132=50.000
$133=300.000
$134=300.000
$135=300.000

GCODE

; LightBurn 1.4.01
; GRBL device profile, current position
; Bounds: X0 Y0 to X80 Y80
G00 G17 G40 G21 G54
G91
M4
; Cut @ 300 mm/min, 15% power
M8
G0 X0Y0
; Layer Labels
G1 Y80S150F300
G1 X80
G1 Y-80
G1 X-80
M9
G1 S0
M5
; return to starting pos
G0 X0Y0
M2

Enzo-Enz, 2023-08-27:
> Premise: with my old genmitsu everything was ok.
>
> A simple square on lightburn. The machine engraves well the first and the second edges. Starts the third one. Somewhere it suddenly stops for a [...]

Hi bro 
I had the same issue before and It was because of lightburn
try another version of lightburn

tjivanhoff, 2023-09-06:
The problem is over. It was the stepper motor controller voltage setting. When I put the correct voltage the engraver goes well.

nebneru85, 2024-03-25:
Hola alguien sabe para que son los parametros  $50 $38 $40