makerbase-mks/MKS-DLC32 #409: Another M4 error with DLC32 V2.1.003

guyjami, 2023-09-06:
Hi everybody.

I have replaced an eleksmana SE with a DLC32 V2.1.003. Everything works fine but the laser. I tried a lot of things from other subjects in the forum and posts on internet but nothing worked for me.

When connected , the DLC returns
[ORIGIN:China]
[PRODUCER:MKS_DLC32]
[DATE:20220905]
[VER:1.1h.2023052701:]
[OPT:VPHSW,15,256]
Target buffer size found
[MSG:Machine Name:MKS DLC32]
[MSG:Mode=AP:SSDI=MKS_DLC:IP=192.168.4.1:MAC=24-DC-C3-0A-11-AD]

On the DLC32 H35 screen :
Board:MKS DLC32 CNC V2.1
V2.3(8M.H35.20230527)
CPU:Freq:240MHz/ T:50.0C/ ID:44049

When i try to start a stream with lightburn or lasergrbl, i get the following message :
[MSG:M4 requires laser mode or a reversable spindle]
error:20
Unsupported or invalid g-code command found in block.

The laser mode is well activated $32=1
Don't known where the problem is ?
Any idea ?
thank's

mv-83, 2023-10-26:
i have same problem,almost certainly the problem is the latest firmware

guyjami, 2023-10-27:
It was in fact a firmware issue.
The V2.3(8M.H35.20230527) provided with the board had issue with m4 command event the laser mode ($32=1) is enabled.
I have flashed the board with an older one (V2.30_H35_2022_1223_N_ZX_001.bin) downloaded here on github :[https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware/TS35/Laser/Normal/Board_V2.0 10](https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware/TS35/Laser/Normal/Board_V2.0)
Now it works perfectly with lightburn and M4 command even via USB and TS35 Screen connected !

guyjami, 2023-10-27:
Finally, i just changed the value $23 from 5 to 3 to have the same origin position than lightburn.
So my old eleksmaker A3 is more silent, capable of working standalone or via wifi, able to drive a rotary module and can do homing via limit switches.

closed 2023-10-27 by guyjami

mv-83, 2023-10-27:
> It was in fact a firmware issue. The V2.3(8M.H35.20230527) provided with the board had issue with m4 command event the laser mode ($32=1) is enabled. I have flashed the board with an older one [...]

bad for me, I had to compile the source code to swap the pins of the Y axis with those of the Z axis (on the board there are 2 connectors for the Y motors, but I have 2 motors in the Z axis) and I don't know where look for the bug

guyjami, 2023-10-29:
you can wire your two Z motors in parallel. That's what i've done on my ender 3 v2 when i have added a second Z motor. It exist cable adapters to do the job.

dhuff0271, 2025-05-17:
I think it's here
https://github.com/makerbase-mks/MKS-DLC32/blob/main/MKS-DLC32-main/firmware/TS35/Laser/Normal/Board_V2.0/V2.30_H35_2022_1223_N_ZX_001.bin