makerbase-mks/MKS-DLC32 #410: Dangerous TTL calculation error - what is the $30 setting for 100% and when will this be fixed

Wiki-the-Tinkerer, 2023-09-08:
Just been measuring TTL output from the 3pin output (mainly for laser, but also used by external VDF:s, as it's supposed to be 0-5v). Discoveries:

**There seems to be a SERIOUS calculation error in the firmware, which is downright dangerous!!** 
This thing could burn houses or take a mans arm off - not informing about this in big letters at the front of all the channels about this board is potentially lethally dangerous.

Example:
I have one machine that has a max of 24000 rpm. So in the grbl settings $30=24000. Now the expectation is that 24000rpm gives out 5v, and 0rpm 0v.

When testing, 0 gives out 0. good. BUT 1000rpm gives out 2.05v, 2000 rpm 4.11v, 3000rpm 1.23v.. and 24000 rpm gives out 0.

So there probably is some "base" assumption rpm value for 100%, where the division works as supposed - at least say what this is as a quick fix.

2) additional comment- the pins are marked unclearly.  The pin order is Signal-Ground-Vin. On the board it says S-TTL-Vin, making it look like the middle pin is the TTL signal. At least an easy mistake to make.

MJ666, 2023-09-08:
The signal is an digital 1 kHz PWM output. The PWM frequency can be changed. Measure the voltage with an DVM will likely not give correct results. You need to check the signal with an Oscilloscope. To control a VFD with an analog voltage signal you need some kind of a converter to transform the PWM to an analog signal.

Wiki-the-Tinkerer, 2023-09-11:
Hi MJ666.. Yes, kinda of course. But.., not really. I'm fairly sure there is an internal pwm calculation error, as the "rough" output should be somewhat following the power sent the spindle (or laser). I think I somewhere saw somewhere a comment about setting  $30 to 1000 (maybe) - but have not been able to find any clear info.

As for the more accurate pwm to analogue, there is a couple good examples here [www.analog.com/en/technical-articles/accurate-fast-settling-analog-voltages-from-digital-pwm-signals.html](https://www.analog.com/en/technical-articles/accurate-fast-settling-analog-voltages-from-digital-pwm-signals.html) for only a few bucks.

MJ666, 2023-09-13:
I'm using the MKS firmware with my Atomstack A20 laser and the PWM signal is working normally. $30 is set to 1000.