makerbase-mks/MKS-DLC32 #416: Hard limit swithes with WiFi connected

SlaUA, 2023-09-15:
Hello, I just installed MKS-DLC32 v2.1 board on a CNC router, connected the hard limit switches (normally closed), so $5=0, enabled them by $21=1, rebooted the board, checked. Everything works fine without WiFi connection. When WiFi is connected, opened=not connected X limit switch gives limit switch error all the time. On the connected screen TS35 appears a window with the message "Hard Limit!" and when I click "Yes" it appears again untill X limit pin is shorted to GND. Y and Z limits work just fine. Tested with firmware V2.0.8_H35_20220105_N_CNC.bin, V2.1.0_H35_20220510_N_CNC_001.bin, V2.1.0_H35_20220526_02_N_CNC.bin - behavior remains the same. Can anyone check this please? Thank you very much.

zebsboards, 2024-07-06:
Did you ever figure this out?  I'm just now having the same issue

zebsboards, 2024-07-07:
Posting this in case anyone else has the same issue in the future.

The fix for me was to increase the homing switch pulloff value ($27) to 2mm from the default value of 1mm

Even though the 1mm value worked fine in every mode except when wifi was enabled, changing the value to 2mm made the difference and I no longer get the hard limit error.

jnl1105, 2024-07-08:
I normally do a 4mm to 5mm pull off but I normally use roller arm style instead of push button.