makerbase-mks/MKS-DLC32 #417: Advice needed: Browser security settings

deonholt, 2023-09-20:
Hi all and thanks for your help.

I am very new to this and just trying to familiarise myself with the Web UI. I have tried Firefox & Google Chrome. None of the browsers want to display the page. It states a warning that I don't have a secure connection. I have tried with my Samsung's Chrome & Safari as well as my Win 10 Chrome & Safari. I have even added the address: 192.168.1.3 as an exception but is still displays an empty page.

Please advise about settings or another browser that doesn't have this issue.

Thanks very much.

SlaUA, 2023-09-20:
Hi, please tell the firmware version and why do you use 192.168.1.3 address?

deonholt, 2023-09-20:
Firmware is: V2.10

I use this address as this is the IP address of the CNC Machine? Am I trying to access the Web UI in the wrong way? How else do I access the Web UI? As I'm saying: I am ver new to this.

Thanks for your help.

[image]

SlaUA, 2023-09-20:
On the screen you showed, tap WiFi, there should be IP address of the board in your network.