makerbase-mks/MKS-DLC32 #420: Rasters are engraving wavy on both x and y axis...

quasarn01, 2023-09-29:
Trying to figure out why my machine with a MKS DLC32 ver. 2.1 board will engrave excellent using vectors, but, when using rasters the x or the y axis overshoots in places and the pic comes out wavy and distorted on the edges…  Below are the pics with one being the raster images and the other two being the vectors...  I've done everything I could think of to solve this issue...  Belts are tight, frame is square, diode module set tight with eccentric screw, etc...  Can anyone point me in the right direction?  Using Lightburn with a LaserTree 10w diode module...
[image]
[image]
[image]

zuber6, 2023-11-03:
I have exactly the same problem. Perfect vectors but raster images come out wavy. Improves with reduction in speed and/or dpi, but still not good. Have tried many things including changing overscan etc. Using lightbirn software.

BaBaika666, 2023-12-14:
Есть решение для mks dlc32?

j46, 2024-01-16:
j'ai exactement le même problème, avez vous trouvé une solution?

zuber6, 2024-01-17:
There is a "solution" in this post but I could not see any improvement.
https://github.com/makerbase-mks/MKS-DLC32/issues/431

quasarn01, 2024-01-17:
Check to see what your value for $0 is...  Mine was set @ $0=6 so I changed it to $0=10 and it drastically improved the output...

zuber6, 2024-01-17:
I'll try it. I have a ramps board set up identically and it works perfectly so it has to be a problem with the mks firmware.

j46, 2024-01-20:
je vient d'essayer le $0=10, et oui ca améliore très nettement la qualité, Merci pour l'info.
de plus je me suis intéressé au $1 qui est le delay de maintient du moteur alimenté a la fin du déplacement pour ne pas que l'axe ce décale avec l'inertie (je pense que c'est ca), moi il était a 25 microsecondes.
j'ai mis $1 a 5.
[image]
en bas a gauche: $0=6 $1=25
a droite : $0=10 $1=25
en haut a gauche $0=10 $1=5 (meilleur qualité)
j'ai fait un autre essai $1 a 255 (255 supprime le delay), le problème est que les moteurs reste en permanence alimenté même après le cycle terminé, donc risque de surchauffe des moteurs.

zuber6, 2024-03-01:
works perfectly! Thanks all!