makerbase-mks/MKS-DLC32 #422: 80w laser doesn't work on dlc32

Alireza-90, 2023-10-04:
Hello 
Recently I used a mks dlc32 for my laser but 80w laser module doesn't work with it 
Does anyone had this problem ?
Any ideas what can be wrong with the module , wiring or anything else?

Plottmania, 2023-10-29:
80 watts?

What kind of laser or what kind of laser module do we have?
A diode laser with three cables: VCC / GND / Signal
Then it depends on whether 12 volts or 24 volts!

If only 12 volts go into the DLC 32, only 12 volts come out for the laser.
If 24 volts go in, the laser should also tolerate 24 volts!
Maybe a 12 volt laser driver can handle 24 volts, but the mounted fan usually can't...then you have a little problem.

The laser mode is set in the GRBL: $32=1 (now the laser is active).

Any questions?

Alireza-90, 2023-10-29:
Thank you plottmania