makerbase-mks/MKS-DLC32 #423: M7 and M9 inverted

mscart, 2023-10-05:
Hi there. 
I try to use the SCL pin to command a 5 V relay to stand and stop air assist. but when i power the board the SCL pin is High and the relay si on. 

when i try the m7 command the relay is off and with m9 the relay is on.  m9 should off the relay and m7 should start. what can i do ? 

i use  this firmware : /TS35/Laser/Normal/Board_V2.0/V2.0.8_H35_20220105_N.bin

Babaika1987, 2023-10-07:
Здравствуйте mscart.
Я решил данную проблему так, собрал схему смотрите картину 
[image]

mscart, 2023-10-08:
thank you ... i will try it

ElectricBluefishProductionsInc, 2023-11-21:
I am having a similar issue. Has anyone found that an optocoupler solves the issue presented above?

mscart, 2023-12-11:
> I am having a similar issue. Has anyone found that an optocoupler solves the issue presented above?

not yet i don't find in romania that integrated circuit.