makerbase-mks/MKS-DLC32 #429: mks dlc32 + 80w diode laser 3 pin 24v,  not work in max power ttl port

redwishi, 2023-10-28:
hi . i build a laser machine with  **80w 24v diode laser**( twotree brand) and buy **mks dlc32 v2** board for control this laser head. and update mks firmware to lasetes V2.30_H35_2022_1223_N_ZX_001 (ts35 , normal laser , v2 board)

the mks dlc32 ttl 3 pin input ,can support max 50w laser at 24v (Bottom image)
[image]

so i select this wiring for use all of the 80w power  (Bottom image)

[image]

but i can **not** use the maximum power of laser . For example based on references in the 80w laser i can cut 3mm plywood with 1 time, But I succeeded after 6 rounds with (100% power and 150mm/min speed)

i choose another wiring method (12-24V-PWM type (2PIN))  (Bottom image)

[image]

this wiring method use 100% of laser power (when the plywood plate under the laser made a hole in 2 sec)
but the problem is the laser module always on Even from the time of turning on the device and can not laser off with ts35 LCD menu.

whats the problem ? whats the solution ؟

danielinkiel, 2023-10-28:
picture 1 plus mosfet

redwishi, 2023-10-28:
> picture 1 plus mosfet

thanks for your answer 
i have a fotek ssr-40 da from my old 3d printer . can i use it ? and how wiring this mosfet  ?

[image]

redwishi, 2023-10-29:
i test picture 1 wiring with 2 power supply and share ground pin with them but the problem still exists . i know the 1 power with 14.6a is enough .

danielinkiel, 2023-10-29:
example
[image]

redwishi, 2023-10-29:
this 

> example [image]

this module for 12v laser . my diode laser is 24v .  the output is 4 wire but my laser have 3 wire  . (ttl G +V)

danielinkiel, 2023-10-29:
The illustration is only intended to explain the function. neje modules for example has a 4 “output” for the temperature sensor (?) which you dont need.
What's important (I think) is that you get the ground off the board

Plottmania, 2023-10-29:
Why should the laser not work?
There is a GRBL on it and in the GRBL the line
$32=0 (laser not active)
$32=1 (laser active)

I connect every laser with 24 volts to this connector: 24 V / GND / Sigal

redwishi, 2023-10-29:
> The illustration is only intended to explain the function. neje modules for example has a 4 “output” for the temperature sensor (?) which you dont need.

i you right 

> What's important (I think) is that you get the ground off the board

Your suggestion is to connect only ttl to the board then ground and v+ connect to the power supply ?

redwishi, 2023-10-29:
> I connect every laser with 24 volts to this connector: 24 V / GND / Sigal

in the mks dlc32 datasheet and wiring PDF says the board can handle maximum 50w laser at 24v >> my laser is 80w at 24v

Plottmania, 2023-10-30:
I don't think your laser is 80 watts if it is a diode laser.
Something else if you have a CO2 laser.
Is it a diode laser that has an 80 watt power consumption?
Possibly, but 80 watts will never come out optically....that would be about 15 diodes with about 6 watts per diode....isn't there yet : )

Again, the driver of the laser doesn't care because it only needs a TTL signal to switch.
So it is 0 -5 Volt (usual up to 2,5 Volt)
So ON and OFF, that's all it is.

This is not magic but a logical circuit.

Your driver now has to make something out of it and the laser starts to shine (the diodes) and the diodes in turn dim or brighten depending on the set power...dim.

Plottmania, 2023-10-30:
The board from Neje can be used where the green cable is for the temperature sensor.
But you can also make your own cable : )
Usual would be:

VCC : 12 / 24 Volt (RED)
GND (BLACK)
Signal (YELLOW)

Depending on the provider, the pin assignments may be swapped, so watch out.

Plottmania, 2023-10-30:
An MKS DLC with external drivers, 12 and 24 volt power supplies and various lasers connected to it.
The times with 12 or 24 volts run...very simple : )

[image]
[image]

redwishi, 2023-10-30:
> er modules mounted and can be changed so quickly

> I don't think your laser is 80 watts if it is a diode laser.
> Something else if you have a CO2 laser.
> Is it a diode laser that has an 80 watt power consumption?
> Possibly, but 80 watts will never come out optically....that would be about 15 diodes with about 6 watts per diode....isn't there yet

thanks for your response

my laser is 80w power consumption and 10w laser power 

[image]